adafruit/Adafruit_CircuitPython_BNO08x #5: Library does not install properly on Raspberry Pi

jerryneedell, 2020-10-19:
I have tried installing this library on a Raspberry Pi4 via pip3 and it says it installs but when I try to use it it reports 
that there is not adafruit_bno8x module

When I look where my python3 modules are stored by pip3, I do not see an folder fo adafruit_bno8x.
there is a folder 
adafruit_circuitpython_bno08x-1.0.0.dist-info
but not the actual module.

i have tried installing  it both via
pip3 install adafruit_circuitpython_bno0x8
and
sudo pip3 install adafruit_circuitpython_bno0x8

the results are the same
pip3 install to  ~/.local/lib/python3.7/dist-packages/
sudo pip3 install to /usr/local/lib/python3.7/dist-packages/

jerryneedell, 2020-10-19:
The bno08x_simpletest.py script runs normally if I run it from a folder with the adafruit_bno08x library in it.

the bno0x8_more_reports.py script fails with this error
```
pi@gjnpi4ram4g:~/projects/blinka/bno08x $ python3 bno08x_more_reports.py 

		********** Packet *************
DBG::		 HEADER:
DBG::		 Data Len: 19
DBG::		 Channel: INPUT_SENSOR_REPORTS (3)
DBG::		 	Report Type: BASE_TIMESTAMP (0xfb)
DBG::		 Sequence number: 170

DBG::		 Data:
DBG::		[0x04] 0xFB 0x47 0x00 0x00 
DBG::		[0x08] 0x00 0xFF 0xFF 0xFF 
DBG::		[0x0C] 0xFF 0xFF 0xFF 0xFF 
DBG::		[0x10] 0xFF 0xFF 0xFF 0xFF 
DBG::		[0x14] 0xFF 0xFF 0xFF 
		*******************************

Traceback (most recent call last):
  File "bno08x_more_reports.py", line 22, in <module>
    bno.enable_feature(adafruit_bno08x.BNO_REPORT_ACTIVITY_CLASSIFIER)
  File "/home/pi/projects/blinka/bno08x/adafruit_bno08x/__init__.py", line 975, in enable_feature
    self._process_available_packets(max_packets=10)
  File "/home/pi/projects/blinka/bno08x/adafruit_bno08x/__init__.py", line 783, in _process_available_packets
    self._handle_packet(new_packet)
  File "/home/pi/projects/blinka/bno08x/adafruit_bno08x/__init__.py", line 841, in _handle_packet
    raise error
  File "/home/pi/projects/blinka/bno08x/adafruit_bno08x/__init__.py", line 836, in _handle_packet
    _separate_batch(packet, self._packet_slices)
  File "/home/pi/projects/blinka/bno08x/adafruit_bno08x/__init__.py", line 351, in _separate_batch
    required_bytes = _report_length(report_id)
  File "/home/pi/projects/blinka/bno08x/adafruit_bno08x/__init__.py", line 342, in _report_length
    return _REPORT_LENGTHS[report_id]
KeyError: 255
```

ladyada, 2020-10-19:
on a raspi you may have to use clockstretching hacks

jerryneedell, 2020-10-19:
Ah- will look into clock stretching.
But there is still something wrong with the pip install...

FoamyGuy, 2020-10-19:
I think the pip install issue is inside the `setup.py` file here: https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/blob/cac28f28159fa0a5bcee29227179ad5a45156ddf/setup.py#L58

I think it needs to be changed to `packages` instead of `py_modules` and it should fix installing via pip.

siddacious, 2020-10-19:
@FoamyGuy Is right; I'll take care of this later today

FoamyGuy, 2020-10-20:
fixed with #6

closed 2020-10-20 by FoamyGuy