adafruit/Adafruit_CircuitPython_BNO08x #8: rendering issue with  `activity_classification` docs

siddacious, 2020-11-04:
Note that the list of activities comes before the text that references it:

[image]

dherrada, 2020-11-25:
I tried building the docs locally, and it built like it was supposed to. However, it seems it is not building like that here. I tried re-running actions after making a tiny change (it had been too long since the last actions run to just re-run it) but that didn't do anything. 
[image]
The same thing is happening with stability_classification, which would lead me to believe it has something to do with the bullet points for whatever reason.

One other thing I noticed is that locally, the light gray "property" thing appears before the name of the property locally, but is totally absent on the online docs.

siddacious, 2020-11-27:
@dherrada this seems like a difference in how Sphinx is being run locally and on readthedocs.org; Could be a Sphinx version difference, or a config file difference, however I believe RTD should be using our included config file.

jposada202020, 2021-04-23:
I build the documents in my readthedocs account and the documents are ok.
https://adafruit-circuitpython-bno08x-jdm.readthedocs.io/en/latest/api.html

We are having a similar issue here
https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/issues/34#issuecomment-825064756

@kmatch98 maybe this is related the way the CI and our local machines are building the documentation

dhalbert, 2021-04-23:
What versions have you installed? Let's see what `pip3 list |egrep 'sphinx|recommonmark'` says, and compare that to the CI settings.

jposada202020, 2021-04-23:
So System wise:
```
recommonmark                             0.7.1               
sphinx-autoapi                           1.7.0               
sphinx-rtd-theme                         0.5.1               
sphinxcontrib-applehelp                  1.0.2               
sphinxcontrib-devhelp                    1.0.2               
sphinxcontrib-htmlhelp                   1.0.3               
sphinxcontrib-jsmath                     1.0.1               
sphinxcontrib-qthelp                     1.0.3               
sphinxcontrib-serializinghtml            1.1.4               
sphinxcontrib-svg2pdfconverter           1.1.1   
```
However, the problem is not in my computer, I can build the documents fine.

For the displayIo library is contained in a ven, so doing the same command came empty. Just for information, I coudl build the documentation and create the html using `sphinx-build -E -W -b html . _build/html` without any problem inside.

jepler, 2021-04-26:
Keep in mind that what the github CI builds is not what RTD builds, that is a separate process.

closed 2021-05-25 by dhalbert