adafruit/Adafruit_CircuitPython_BNO08x #12: i2c cannot assign attribute direction

dheera, 2020-12-29:
FeatherS2
CircuitPython 6.x

```
import board
import busio
import adafruit_bno08x

i2c = busio.I2C(board.SCL, board.SDA)
bno = adafruit_bno08x.BNO08X(i2c)
```

output:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_bno08x/__init__.py", line 508, in __init__
  File "adafruit_bno08x/__init__.py", line 513, in initialize
  File "adafruit_bno08x/__init__.py", line 1043, in hard_reset
AttributeError: 'I2C' object cannot assign attribute 'direction'
```

tannewt, 2020-12-30:
I think you need to use the I2C version of the class: https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/blob/master/adafruit_bno08x/i2c.py#L16

tannewt, 2021-01-05:
@dheera Did that fix it?

jposada202020, 2021-05-11:
@dheera  HAve you been able to test? Thanks

dheera, 2021-05-14:
Yes, that fixed it. Will close issue. Thanks!
For anyone who lands on this page, this is how you use it:
```
import board
import busio
import time
import adafruit_bno08x
from adafruit_bno08x.i2c import BNO08X_I2C
i2c = busio.I2C(board.SCL, board.SDA, frequency=800000)

bno = BNO08X_I2C(i2c)
bno.enable_feature(adafruit_bno08x.BNO_REPORT_ACCELEROMETER)
bno.enable_feature(adafruit_bno08x.BNO_REPORT_GYROSCOPE)
bno.enable_feature(adafruit_bno08x.BNO_REPORT_MAGNETOMETER)
bno.enable_feature(adafruit_bno08x.BNO_REPORT_ROTATION_VECTOR)

while True:
      print("acc", bno.acceleration)
      print("gyr", bno.gyro)
      print("mag", bno.magnetic)
      print("quat", bno.quaternion)
      time.sleep(0.1)
```

closed 2021-05-14 by dheera