adafruit/Adafruit_CircuitPython_BNO08x #16: Library throws key error exception with Learning Guide example code

hukuzatuna, 2021-02-03:
I have a freshly-configured Raspberry Pi 4 (8GB) that generates an exception when exercising the BNO085 sensor. When I first tried it I was using the Beta 64-bit OS, so I thought it was a storage boundary problem. Reverted to the standard 32-bit OS and the problem persists. Configured a Raspberry Pi Pico and attached the same device, and it ran flawlessly for days (ruling out a hardware problem with the sensor?)

Problem:
```
pi@matchbox:~ $ bno085

		********** Packet *************
DBG::		 HEADER:
DBG::		 Data Len: 15
DBG::		 Channel: INPUT_SENSOR_REPORTS (3)
DBG::		 	** UNKNOWN Report Type **: 0x7b
DBG::		 Sequence number: 14

DBG::		 Data:
DBG::		[0x04] 0x7B 0x07 0x00 0x00 
DBG::		[0x08] 0x00 0x01 0x04 0x02 
DBG::		[0x0C] 0x00 0x9E 0xFF 0xB2 
DBG::		[0x10] 0xF6 0x86 0x03 
		*******************************

Traceback (most recent call last):
  File "/home/pi/bin/bno085", line 22, in <module>
    bno.enable_feature(BNO_REPORT_MAGNETOMETER)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bno08x/__init__.py", line 977, in enable_feature
    self._process_available_packets(max_packets=10)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bno08x/__init__.py", line 785, in _process_available_packets
    self._handle_packet(new_packet)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bno08x/__init__.py", line 843, in _handle_packet
    raise error
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bno08x/__init__.py", line 838, in _handle_packet
    _separate_batch(packet, self._packet_slices)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bno08x/__init__.py", line 353, in _separate_batch
    required_bytes = _report_length(report_id)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bno08x/__init__.py", line 342, in _report_length
    return _AVAIL_SENSOR_REPORTS[report_id][2]
KeyError: 123
pi@matchbox:~ $ 
```
Code that generated the problem (taken straight from the learning guide):
```
#!/usr/bin/python3

# SPDX-FileCopyrightText: 2020 Bryan Siepert, written for Adafruit Industries
#
# SPDX-License-Identifier: Unlicense
import time
import board
import busio
from adafruit_bno08x import (
    BNO_REPORT_ACCELEROMETER,
    BNO_REPORT_GYROSCOPE,
    BNO_REPORT_MAGNETOMETER,
    BNO_REPORT_ROTATION_VECTOR,
)
from adafruit_bno08x.i2c import BNO08X_I2C

i2c = busio.I2C(board.SCL, board.SDA, frequency=800000)
bno = BNO08X_I2C(i2c)

bno.enable_feature(BNO_REPORT_ACCELEROMETER)
bno.enable_feature(BNO_REPORT_GYROSCOPE)
bno.enable_feature(BNO_REPORT_MAGNETOMETER)
bno.enable_feature(BNO_REPORT_ROTATION_VECTOR)

while True:

    time.sleep(0.5)
    print("Acceleration:")
    accel_x, accel_y, accel_z = bno.acceleration  # pylint:disable=no-member
    print("X: %0.6f  Y: %0.6f Z: %0.6f  m/s^2" % (accel_x, accel_y, accel_z))
    print("")

    print("Gyro:")
    gyro_x, gyro_y, gyro_z = bno.gyro  # pylint:disable=no-member
    print("X: %0.6f  Y: %0.6f Z: %0.6f rads/s" % (gyro_x, gyro_y, gyro_z))
    print("")

    print("Magnetometer:")
    mag_x, mag_y, mag_z = bno.magnetic  # pylint:disable=no-member
    print("X: %0.6f  Y: %0.6f Z: %0.6f uT" % (mag_x, mag_y, mag_z))
    print("")

    print("Rotation Vector Quaternion:")
    quat_i, quat_j, quat_k, quat_real = bno.quaternion  # pylint:disable=no-member
    print(
        "I: %0.6f  J: %0.6f K: %0.6f  Real: %0.6f" % (quat_i, quat_j, quat_k, quat_real)
    )
    print("")
```
Expected behavior:
It should print the IMU data.

caternuson, 2021-02-03:
Related?
https://forums.adafruit.com/viewtopic.php?f=19&t=174424

Dealing with clock stretching is always the first suspicion. But the BNO08x seems to be ever more "fun".

hukuzatuna, 2021-02-03:
> Related?
> https://forums.adafruit.com/viewtopic.php?f=19&t=174424
> 
> Dealing with clock stretching is always the first suspicion. But the BNO08x seems to be ever more "fun".

Yes, clearly related! I had tried the 10kHz baudrate trick, but it does not work. As per the forum suggestion I added
```
dtparam=i2c_arm_baudrate=400000
```
to /boot/config.txt, rebooted, and the sensor is working as expected. Thank you @caternuson ! (Closing issue)

closed 2021-02-03 by hukuzatuna

ladyada, 2021-02-03:
@caternuson ok want to add to guide?

caternuson, 2021-02-03:
I'll update it for now, since this seems to work. And the 10kHz slowdown does not. But seems really weird to me that speeding up the I2C clock is fixing a clock stretching issue. I'm concerned there's more going on with this sensor maybe.

ladyada, 2021-02-03:
maybe at too-low rates it cant read the reports fast enough, and the sensors gets 'backed up'?