adafruit/Adafruit_CircuitPython_BNO08x #28: Not working with I2C on MCP2221

caternuson, 2021-07-29:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=181641

Tried various slower bus speeds and still no success.

truher, 2024-07-10:
this combination doesn't work for me either.

dhalbert, 2024-07-10:
The BNO085 is flaky enough that I'm not surprised: https://learn.adafruit.com/i2c-addresses/troublesome-chips

truher, 2024-07-10:
i'm curious if anyone has had success with the UART mode between these two modules?

bhupiister, 2024-07-22:
I am testing the same right now and even i get this error. @ladyada Any thoughts on this?

ladyada, 2024-07-22:
BNO085 I2C is kinda a mess, i'd try the RVC UART mode

dhalbert, 2025-05-29:
You could try the stronger pullup on SDA trick mentioned here: https://forums.adafruit.com/viewtopic.php?t=183941, but it may not help. Let us know if it does. As [mentioned](https://learn.adafruit.com/i2c-addresses/troublesome-chips), the chip violates various I2C protocol rules. You could try the UART RVC mode, as suggested, or connect the chip to a microcontroller that does work with it and communicate with the microcontroller over serial

jknecht-chromo, 2025-05-30:
@dhalbert, thanks for the tip — I’ll give the stronger SDA pull-up trick a try and report back if it moves the needle.

I’m also eyeing the Adafruit breakout (product 4970) to see if it sidesteps the MCP2221A’s I²C quirks — have you ever used that board? Ideally I’d like to stick with a USB interface, but if it keeps misbehaving I’ll fall back to slinging it through a microcontroller over serial. Any experience or thoughts on the breakout would be super helpful!

jknecht-chromo, 2025-06-05:
Hello @dhalbert, I have switched to the sparkfun FT232X board to switch to a UART connection:
https://www.sparkfun.com/sparkfun-ft231x-breakout.html

 I am still encountering the error. I have been able to narrow it down to asking for the calibration status causes this to happen. 
When I ask for magnetic and quaternion, I do not encounter this error. Once I add the status check, then I encounter this error.

jknecht-chromo, 2025-06-10:
@dhalbert  I think I was able to resolve this problem. I think it is related to constantly checking the calibration status. 
        #mag_status = self.bno.calibration_status & 0x03 #only used for startup checking 
        mag_status = self.bno._magnetometer_accuracy

bno.calibration_status internally sends a ME Command (Management Entity request) to the BNO08X over the SHTP protocol.
This involves a special packet exchange and can fail intermittently over UART due to timing, packet sync issues, or congestion — especially when reading other features rapidly.

On the other hand, bno._magnetometer_accuracy simply reads the last reported accuracy value already embedded in the magnetometer or rotation vector report.
It does not send any additional commands, making it much faster and more robust, especially in high-speed data loops.