adafruit/Adafruit_CircuitPython_BNO08x #30: Missing Type Annotations

FoamyGuy, 2021-09-27:
There are missing type annotations for some functions in this library.

The `typing` module does not exist on CircuitPython devices so the import needs to be wrapped in try/except to catch the error for missing import. There is an example of how that is done here:

```python
try:
    from typing import List, Tuple
except ImportError:
    pass
```

Once imported the typing annotations for the argument type(s), and return type(s) can be added to the function signature. Here is an example of a function that has had this done already:

```python
def wrap_text_to_pixels(
    string: str, max_width: int, font=None, indent0: str = "", indent1: str = ""
) -> List[str]:
```
If you are new to Git or Github we have a guide about contributing to our projects here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

There is also a guide that covers our CI utilities and how to run them locally to ensure they will pass in Github Actions here: https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/check-your-code In particular the pages: `Sharing docs on ReadTheDocs` and `Check your code with pre-commit` contain the tools to install and commands to run locally to run the checks.

If you are attempting to resolve this issue and need help, you can post a comment on this issue and tag both @foamyguy and @kattni or reach out to us on Discord: https://adafru.it/discord in the `#circuitpython-dev` channel.

The following locations are reported by mypy to be missing type annotations:

- [ ] adafruit_bno08x/\_\_init\_\_.py:208
- [ ] adafruit_bno08x/\_\_init\_\_.py:213
- [ ] adafruit_bno08x/\_\_init\_\_.py:237
- [ ] adafruit_bno08x/\_\_init\_\_.py:241
- [ ] adafruit_bno08x/\_\_init\_\_.py:255
- [ ] adafruit_bno08x/\_\_init\_\_.py:266
- [ ] adafruit_bno08x/\_\_init\_\_.py:294
- [ ] adafruit_bno08x/\_\_init\_\_.py:299
- [ ] adafruit_bno08x/\_\_init\_\_.py:313
- [ ] adafruit_bno08x/\_\_init\_\_.py:328
- [ ] adafruit_bno08x/\_\_init\_\_.py:348
- [ ] adafruit_bno08x/\_\_init\_\_.py:355
- [ ] adafruit_bno08x/\_\_init\_\_.py:388
- [ ] adafruit_bno08x/\_\_init\_\_.py:462
- [ ] adafruit_bno08x/\_\_init\_\_.py:476
- [ ] adafruit_bno08x/\_\_init\_\_.py:497
- [ ] adafruit_bno08x/\_\_init\_\_.py:750
- [ ] adafruit_bno08x/\_\_init\_\_.py:787
- [ ] adafruit_bno08x/\_\_init\_\_.py:805
- [ ] adafruit_bno08x/\_\_init\_\_.py:830
- [ ] adafruit_bno08x/\_\_init\_\_.py:842
- [ ] adafruit_bno08x/\_\_init\_\_.py:847
- [ ] adafruit_bno08x/\_\_init\_\_.py:857
- [ ] adafruit_bno08x/\_\_init\_\_.py:881
- [ ] adafruit_bno08x/\_\_init\_\_.py:904
- [ ] adafruit_bno08x/\_\_init\_\_.py:952
- [ ] adafruit_bno08x/\_\_init\_\_.py:966
- [ ] adafruit_bno08x/\_\_init\_\_.py:1035
- [ ] adafruit_bno08x/\_\_init\_\_.py:1039
- [ ] adafruit_bno08x/\_\_init\_\_.py:1082
- [ ] adafruit_bno08x/\_\_init\_\_.py:1088
- [ ] adafruit_bno08x/\_\_init\_\_.py:1092
- [ ] adafruit_bno08x/uart.py:68
- [ ] adafruit_bno08x/uart.py:72
- [ ] adafruit_bno08x/uart.py:111
- [ ] adafruit_bno08x/spi.py:67
- [ ] adafruit_bno08x/spi.py:119
- [ ] adafruit_bno08x/spi.py:172
- [ ] adafruit_bno08x/spi.py:185
- [ ] adafruit_bno08x/i2c.py:64
- [ ] adafruit_bno08x/i2c.py:70
- [ ] adafruit_bno08x/i2c.py:133

rrahkola, 2023-04-25:
Picking this up...

rrahkola, 2023-04-26:
Hi @FoamyGuy, opened a [PR (#42)](https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/pull/42) for this.