adafruit/Adafruit_CircuitPython_BNO08x #33: No pull up found on SDA or SCL when using Feather RP2040 with example/bno08x_simpletest.py

dwitt, 2021-11-14:
I cannot get the simpltest.py example to work with the Feather RP2040. I have tried several feathers and a section BNO08X. I have had this board work fine with both a Feather M4 and Feather M4 CAN. As far as I know the BNO08X has pull ups. I have tried this both on a breadboard as well as using the STEMMA interface. Any suggestions? I have used the I2C from the Feather RP2040 successfully with other devices such as Honeywell pressure sensors.

ladyada, 2021-11-15:
probably a clockstretch thingy tweak for RP2040 - the BNO080 is kinda terrible in its i2c implemenation
cc @dhalbert

dhalbert, 2021-11-15:
In the meantime, try using `bitbangio.I2C(board.SCL, board.SDA)`, which is a software implementation of I2C, and see if it works better. It may not!

dwitt, 2021-11-15:
I tried using `bitbangio.I2C(board.SCL, board.SDA)`, but this did not improve things. I did get an error about clock stretch so I tried extending the timeout to 1000 using `bitbangio.I2C(board.SCL, board.SDA, timeout = 1000)`. This worked!

Finally, I switched back to `busio.I2C(board.SCL, board.SDA, timeout = 1000)` and this also worked. I remembered reading something about bitbangio being used in certain cases with small amounts of data on the RP2040.  With some experimentation I was able to reduce the time out to 500 and still have the I2C work. I was not able to get the I2C to work with a timeout of 255 which I think is the default for bitbangio. 

I found that once I had too short of a timeout the BNO08x would not recover by just changing the timeout and I ended up power cycling it. It's possible that a reset would also work but I did not test that.

caternuson, 2022-10-28:
The "no pull ups found" message is most commonly some kind of connection issue. Either due to wiring or specifying an incorrect I2C port (to which nothing is attached).

Just tested with a BNO085 connected via a STEMMA QT cable and can not reproduce:
```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather RP2040 with rp2040
>>> import board
>>> from adafruit_bno08x.i2c import BNO08X_I2C
>>> bno = BNO08X_I2C(board.STEMMA_I2C())
>>> 
```
@dwitt If you're still monitoring this issue thread, can you try the above simple check and see if you are still getting the "no pull ups" message.

jposada202020, 2023-03-08:
Just to add, I was doing the test for PR #39 . And the simpletest worked for me using I2C.

``` python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Adafruit Feather RP2040 with rp2040
>>> import c
Acceleration:
X: -0.496094  Y: 2.375000 Z: -9.578125  m/s^2

Gyro:
X: 0.000000  Y: 0.001953 Z: -0.001953 rads/s

Magnetometer:
X: -19.250000  Y: 5.875000 Z: -28.125000 uT

Rotation Vector Quaternion:
I: 0.031494  J: 0.991333 K: 0.122009  Real: 0.036682
```

FoamyGuy, 2024-11-12:
Closing this. It appears it was resolved by using the `timeout` argument. Please feel free to re-open or create a new issue if  still facing this problem

closed 2024-11-12 by FoamyGuy