adafruit/Adafruit_CircuitPython_BNO08x #37: Has SPI mode been tested?

caternuson, 2022-10-28:
Re this thread:
https://forums.adafruit.com/viewtopic.php?p=946036#p946036

The guide currently does not cover SPI usage.
https://learn.adafruit.com/adafruit-9-dof-orientation-imu-fusion-breakout-bno085/python-circuitpython

The library example has some code that looks copy-pasta'd (D9 ref'd twice, reset pin set as input) and the instantiation does not match the actual [ctor](https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/blob/c8c7cf01199c658db83e00266f84bb13eb3bab99/adafruit_bno08x/i2c.py#L24):
https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/blob/c8c7cf01199c658db83e00266f84bb13eb3bab99/examples/bno08x_simpletest_spi.py#L19-L25

Additionally, there does not appear to be any code in place to perform the required PS0/Wake pin assertion needed for SPI transactions:
[image]

adam-rutkowski, 2023-02-13:
I see the same problem. When is it going to be fixed? I2C is hopelessly unreliable with this sensor and SPI software support is apparently non existent on Raspbery, it's hugely disappointing.

juuuuulian, 2023-05-10:
Has anyone found a fix for this?

tomy983, 2023-09-03:
`bno = BNO08X_SPI(spi, cs, int_pin, reset_pin, 1000000, debug=True)` 
this gets you just a bit further. All the rest is still not working.
It looks like the SPI implementation is completely broken.

JeroenDelcour, 2023-10-29:
I was bummed to find out about this, because I bought this chip specifically for its SPI support. So I tried to fix it. I got a little further, but I'm stuck.

I added the SPI wake operation like so:

```python
    def soft_reset(self):
        """Reset the sensor to an initial unconfigured state"""
        self._dbg("Soft resetting...", end="")
        data = bytearray(1)
        data[0] = 1
        _seq = self._send_packet(BNO_CHANNEL_EXE, data)
        time.sleep(0.5)

        for _i in range(3):
            try:
                _packet = self._read_packet()
            except PacketError:
                time.sleep(0.5)
        # wake up sensor
        self._wakepin.value = False
        self._wait_for_int()
        self._dbg("OK!")
        # all is good!
```

That seems to fix the initialization; reading the ID from the sensor worked. But doing any more reads gave me a packet with the continuation bit (0x80) set in the header's length field ([see datasheet section 1.3.1](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf)). I'm not sure where that packet is coming from or how to handle it. In the current code, it's labeled as a "half packet" and it just throws a PacketError. I tried ignoring that packet, but I can't seem to get any more packets after that.

Maybe [the Arduino version](https://github.com/adafruit/Adafruit_BNO08x/) can provide some inspiration to someone who is better at C++ than I am.

rotellini9809, 2024-01-10:
hi guys is a shame that spi support doesn't work on software,i hope they fix it soon

tomy983, 2024-01-10:
Hi @rotellini9809 and @JeroenDelcour ,
please check out my issue on the sparkfun library, I posted some code that works with spi, but had some issues with axis swapping while using spi while testing 2 bno086 (but not with i2c). Maybe you could let me know if you also have the same problem.

rotellini9809, 2024-01-11:
> hey hello i'm not super expert what is axis swapping?
>
> Il giorno mer 10 gen 2024 alle ore 22:05 Thomas Sosio <

that axes x is written in axes y for example?

tomy983, 2024-01-11:
I am unable to explain it clearly than I already did in the post I have linked 3 comments up...

dbell37, 2024-07-16:
Oh no, this is still broken in 2024? I started with i2c, but I'm trying to use two sensors (I set one to the alternate address and the other was kept on the stock address: 0x4B 0x4A), but unfortunately the CircuitPython code only supports one sensor. I modified the code to support adding two sensors on i2c, but it hangs at the i2c instantiation of the second sensor.
So I'm therefore trying to use SPI, but that also doesn't work. 
I'll tear everything apart and try serial next. ;)

j-wanner, 2024-08-13:
Has anybody achieved any progress in getting the BNO086 to run with SPI on a Raspberry pi? Are there any other available codebases that you could share?

Would be amazing, I'd really like to have the SPI option to increase the readout rates...

norbinz, 2024-08-31:
Hi j-wanner,
In my experience with BNO085 in the Arduino environment, I don't see much performance difference between I2C and SPI. Also, increasing the bit rate doesn't help much. The report rate bottleneck seems to the BNO's internal processing.

I2C experiment: https://forums.adafruit.com/viewtopic.php?p=857930#p857930
SPI experiment: https://forums.adafruit.com/viewtopic.php?p=859651#p859651

WestJeremy, 2024-09-18:
I have just found that the git repo is out of date for this device. You can go to the [circuit python website](https://circuitpython.org/libraries) and download their 8xx bundles version and there is an up to date BNO08x example that should work on raspberry pi. In that library, they include the reset pin in the class definition of the BNO08x in the SPI.py file. 

Here is the bundle version I am working off of. 

[adafruit-circuitpython-bundle-8.x-mpy-20240917.zip](https://github.com/user-attachments/files/17051116/adafruit-circuitpython-bundle-8.x-mpy-20240917.zip)

georgea96, 2025-02-28:
This still doesn't work for me. I found that in the class of BNO08x there is a TODO saying the sequence_number list is wrong and that it should be one per channel per direction. Also when I wire up everything correctly I do get an IndexError from the sequence list saying basically that the list assignment is not possible. Anyone has an idea what's going on?

```
  File "/home/..../test_spi.py", line 29, in <module>
    bno = BNO08X_SPI(spi,cs,int_pin, reset_pin, debug=True, baudrate=3000000)
  File "/home/.../lib/python3.9/site-packages/adafruit_bno08x/spi.py", line 40, in __init__
    super().__init__(resetpin, debug)
  File "/home/...//lib/python3.9/site-packages/adafruit_bno08x/__init__.py", line 517, in __init__
    self.initialize()
  File "/home/.../imu_env2/lib/python3.9/site-packages/adafruit_bno08x/__init__.py", line 522, in initialize
    self.hard_reset()
  File "/home/.../.../imu_env2/lib/python3.9/site-packages/adafruit_bno08x/spi.py", line 56, in hard_reset
    self._read_packet()
  File "/home/.../.../imu_env2/lib/python3.9/site-packages/adafruit_bno08x/spi.py", line 114, in _read_packet
    self._sequence_number[channel_number] = sequence_number
IndexError: list assignment index out of range
Exiting... 
Cleaning up pins
```