adafruit/Adafruit_CircuitPython_BNO08x #43: Cannot run on top of micropython

Zzx-J, 2023-05-28:
Hi I just followed the instruction on https://learn.adafruit.com/circuitpython-libraries-on-micropython-using-the-raspberry-pi-pico/overview

I have installed all the libraries needed according to the instruction,  when running the example code the IDE pops up with the following issue:

////////
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "adafruit_bno08x/i2c.py", line 27, in __init__
  File "adafruit_bno08x/__init__.py", line 506, in __init__
  File "adafruit_bno08x/__init__.py", line 512, in initialize
  File "adafruit_bno08x/__init__.py", line 1063, in soft_reset
  File "adafruit_bno08x/i2c.py", line 38, in _send_packet
  File "adafruit_bno08x/__init__.py", line 380, in __init__
  File "adafruit_bno08x/__init__.py", line 456, in header_from_buffer
TypeError: function doesn't take keyword arguments
////////

The code I used:

/////////
import board
import busio
from adafruit_bno08x.i2c import BNO08X_I2C
from adafruit_bno08x import BNO_REPORT_ACCELEROMETER

i2c = busio.I2C(scl=board.GP15, sda = board.GP14)
bno = BNO08X_I2C(i2c)
bno.enable_feature(BNO_REPORT_ACCELEROMETER)

while True:
    accel_x, accel_y, accel_z = bno.acceleration  # pylint:disable=no-member
    print("X: %0.6f  Y: %0.6f Z: %0.6f  m/s^2" % (accel_x, accel_y, accel_z))

//////////

I have tried the bme280 library provided by the instruction and it works, which I believe proved that I have correctly installed the library. The reason that I need to use micropython is that I want to use the multicore function. 

Anybody please help.

tekktrik, 2023-05-29:
It looks like MicroPython implements `struct.unpack_from()` to only take positional arguments:  https://docs.micropython.org/en/latest/library/struct.html#struct.unpack_from

Uses of it in this library will need to provide it positionally only in order to keep compatibility.  If you're up for submitting a pull request to fix it, we'd be happy to take a look!

dobodu, 2023-06-09:
Hi [Zzx-J](https://github.com/Zzx-J) and [tekktrik](https://github.com/tekktrik)

I'm in the same case,
I adapted the CircuitPython Library to Micropython in this repo [https://github.com/dobodu/Adafruit_MicroPython_BNO08x][(url](https://github.com/dobodu/Adafruit_MicroPython_BNO08x))

Some things seems to work (initialisation, can request and read the BNO ID, and many things) but I lack debugging skills to finish the project (I'm struggling for 2 weeks). Obviously I miss something.

If one can help...

tekktrik, 2023-06-12:
I can take a look this week

dobodu, 2023-06-12:
Thank you Alec,

I believe the micropython library for BNO085 is working correctly now.
Nevertheless, I might need testers to check if everything is working fine.

http://github.com/dobodu/Adafruit_MicroPython_BNO08x‌‌

Cheers
De : "Alec Delaney"
A : "adafruit/Adafruit_CircuitPython_BNO08x" ,"dobodu" ,"Comment"
Envoyé: lundi 12 Juin 2023 15:05
Objet : Re: [adafruit/Adafruit_CircuitPython_BNO08x] Cannot run on top of micropython (Issue #43)
 

 

I can take a look this week