adafruit/Adafruit_CircuitPython_BNO08x #46: UART bardrate wrong in examples

PaddyCube, 2023-10-25:
According to your documentation here 
https://learn.adafruit.com/adafruit-9-dof-orientation-imu-fusion-breakout-bno085/python-circuitpython

the baud rate should be set to 115200

```import adafruit_bno08x
from adafruit_bno08x.uart import BNO08X_UART

import serial
uart = serial.Serial("/dev/serial0", 115200)

bno = BNO08X_UART(uart)
```

but this doesn't work. After digging some time, I found that you use a baud rate of 3,000,000 i the examples of the circuit python library.

So I would ask you to change your valuable and clear instructions to the correct value

jason-jaia, 2023-11-22:
We found this as well.

Tekunalogy, 2023-11-23:
fixed in #48

FoamyGuy, 2024-11-12:
resolved by #48

closed 2024-11-12 by FoamyGuy