adafruit/Adafruit_CircuitPython_BNO08x #49: Unprocessable Batch bytes

hopwoodc, 2024-02-02:
I am trying to get a BNO086 working. I'm getting this error:
```
Adafruit CircuitPython 8.2.9 on 2023-12-06; Raspberry Pi Pico with rp2040
>>> 
>>> import board
>>> import busio
>>> from adafruit_bno08x.i2c import BNO08X_I2C
>>> from adafruit_bno08x import BNO_REPORT_ACCELEROMETER
>>> import board
>>> scl=board.GP17
>>> sda=board.GP16
>>> i2c=busio.I2C(scl,sda)
>>> bno = BNO08X_I2C(i2c, address=0x4b)
>>> bno.enable_feature(BNO_REPORT_ACCELEROMETER)

                ********** Packet *************
DBG::            HEADER:
DBG::            Data Len: 2
DBG::            Channel: SHTP_COMMAND (0)
DBG::            Sequence number: 4

DBG::            Data:
DBG::           [0x04] 0x01 0x0E 
                *******************************

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_bno08x/__init__.py", line 999, in enable_feature
  File "adafruit_bno08x/__init__.py", line 803, in _process_available_packets
  File "adafruit_bno08x/__init__.py", line 863, in _handle_packet
  File "adafruit_bno08x/__init__.py", line 858, in _handle_packet
  File "adafruit_bno08x/__init__.py", line 379, in _separate_batch
RuntimeError: ('Unprocessable Batch bytes', 2)
```

I'm able to read data every couple of times, but am often getting error messages like this.

rohangarg2001, 2024-02-18:
Getting the same error while using BNO086

BentPen, 2024-03-14:
Still having this issue as well. BNO085, with Raspberry Pi 3B+.

ReedTruax, 2024-03-18:
I am also having this issue with a PI Pico

BentPen, 2024-03-18:
@ReedTruax I ended up giving up on the BNO085 and bought a LSM9DS1, which worked very nicely. Granted, I didn't use the Adafruit code, but rather the code from this other repo here: https://github.com/Richard-Gemmell/lsm9ds1-rjg. I have no opinion on the Adafruit code, but saw that in his repo Richard Gemmell took the time to incorporate all the data sheet functionality (in particular, sending special bytes to registers to control ODR--output data rate). His work was a goldmine. (Though if you decide to go the same route as I did, take a brief look at the issue I made there.)

diplodocuslongus, 2024-04-30:
> Still having this issue as well. BNO085, with Raspberry Pi 3B+.

For me it's a BNO086, same RPi 3B+, bookworm. No reading so far.

Andrew123098, 2024-08-08:
Any solutions? I am having the same issue with a BNO086 using a Raspberry Pi Zero 2 W

Nate711, 2024-10-22:
@siddacious

praked, 2025-02-26:
Same Problem here, using a BNO086

Andrew123098, 2025-02-26:
> Same Problem here, using a BNO086

Did you try the partial solution I gave above? It works quite well

praked, 2025-02-26:
> > Same Problem here, using a BNO086
> 
> Did you try the partial solution I gave above? It works quite well

I just wanted to register my problem here, I saw your partial work around, would give it a try in the morning. Have you has a look at this https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/i2c-clock-stretching

Andrew123098, 2025-02-26:
> I just wanted to register my problem here, I saw your partial work around, would give it a try in the morning. Have you has a look at this [...]

I couldn't use this workaround because I needed the data as fast as possible. But you can try it. I will say I tried with many different baud rates and saw no difference, though I never went that low.