adafruit/Adafruit_CircuitPython_BNO08x #53: BNO085 and ESP32-S3 unstable I2C address

joseagraz, 2024-12-09:
I'm using the ESP32-S3 TFT Feather and the 9-DOF Orientation IMU Fusion Breakout (BNO085) with CircuitPython and a STEMMA QT cable. When connecting the devices, the default I2C address (0x4A) is not consistently recognized. An I2C scan often shows 0x09, no detection, or rarely 0x4A.

I tried adding a 2K pull-up resistor to SDA, but it didn't solve the issue [reference](https://forums.adafruit.com/viewtopic.php?p=982852&hilit=bno085+ESP32+S3#p982852)

Is there a known solution for this problem? If not, what microcontroller would you recommend as a better fit for this sensor? 
Any advice is greatly appreciated!

dhalbert, 2024-12-09:
Did you see this page? https://learn.adafruit.com/i2c-addresses/troublesome-chips#troublesome-chips-3119905

Do you need wifi? If so try an ESP32-S**2**, instead of S3. If wifi is not needed, try a chip that is not one of 
> Does not work well on i.MX RT10xx chips, ESP32, ESP32-S3.
as listed, such as a SAMD, RP2040, etc.

bradcar, 2025-11-06:
Unfortunately, The BNO080, BNO085, and BNO086 all use **_non-standard clock stretching_** on the i2c. This can cause a variety of issues including report errors and the need to restart sensor. Clock stretching interferes with various chips (ex: RP2) in different ways. If you see sporadic issues this may be part of the issue (BNO08X Datasheet 1000-3927 v1.17, page 15).

I've heard some esp32's can do clock stretching pretty well if enabled right, but others can't.

chipko, 2025-11-07:
Ok, I found something that makes the bno086 work much better on an esp32 .. 

Use software defined i2c pins, not the hardware i2c pins! The software is better suited to clock stretching than the hardware i2c.  I had all sorts of issues but now the worst I get is potentially not recognised on boot but that's ready to manage.

A great sensor and hopefully this helps someone else